iam4722202468/ThinkpadBattery #3: Problems verifying .ino sketch

develroo, 2020-10-24:
Hi. First I want to say thanks very much for this project because it fits my need almost perfectly. I am trying to emulate a battery to power a laptop by a DC power supply. One you have a suitable resistance for Thermal overload it will power on and stay on so long as the OS is not fussy about battery status and power down thinking the battery is broken. This is where your project comes in.

However I downloaded your batteryEmulatior.ino but it fails to verify or compile

```
Arduino: 1.8.13 (Linux), Board: "Digispark Pro (Default 16 Mhz)"

/home/user/devel/git/ThinkpadBattery/batteryEmulator/batteryEmulator.ino: In function 'void setup()':
batteryEmulator:523:19: error: no matching function for call to 'USI_TWI::begin(int)'
   Wire.begin (0x0B);
                   ^
/home/user/devel/git/ThinkpadBattery/batteryEmulator/batteryEmulator.ino:523:19: note: candidate is:
In file included from /home/user/devel/git/ThinkpadBattery/batteryEmulator/batteryEmulator.ino:1:0:
/home/user/.arduino15/packages/digistump/hardware/avr/1.6.7/libraries/Wire/Wire.h:57:13: note: void USI_TWI::begin()
     void    begin();
             ^
/home/user/.arduino15/packages/digistump/hardware/avr/1.6.7/libraries/Wire/Wire.h:57:13: note:   candidate expects 0 arguments, 1 provided
batteryEmulator:524:8: error: 'class USI_TWI' has no member named 'onReceive'
   Wire.onReceive (receiveEvent);
        ^
batteryEmulator:525:8: error: 'class USI_TWI' has no member named 'onRequest'
   Wire.onRequest (requestEvent);
        ^
exit status 1
no matching function for call to 'USI_TWI::begin(int)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

I don't know much about you environment but am I missing something in mine?

Thanks

digmorepaka, 2020-10-24:
try using wire.h only and attinycore instead of digistump. i had this same issue and that resolved it

iam4722202468, 2020-10-30:
digmorepaka is correct, based on your error log it looks like you need to use attinycore.

closed 2020-10-30 by iam4722202468

develroo, 2020-11-03:
Yeah fixed and I compiled the code alas it was not a drop in replacement for my battery/laptop. I don't have the time to reverse engineer the protocol unless you can do it with the batter alone because I have already seconded the laptop into it's 1U case and put it in the cabinet. I can still boot Linux so long as I use `acip=off` which is ok but that also affects the fans if I am not very much mistaken so. 

Wish I had the time and tools to debug the communication but alas I have neither atm. 

Thanks anyway it is a great resource and I will think about getting a data analyser anyway just for the next time I run into something like this.

Cheers!