iam4722202468/ThinkpadBattery #4: universal battery emulator?

5ch4um1, 2020-12-26:
Hello young gentleman! 
First of all, thank you very much for your excellent work!
I was wondering if you realize that your sketch is basically an implementation of the SBS for arduino? I mean, Thinkpads are by far not the only devices that use TI controllers, and your sketch should be really easy to adapt to other arduino platforms and most important: other batteries and devices!
I did some first tests today, it seems to crash when you ask for and address that is not defined, but most of the things my laptop wants to know are already defined. btw, any interest in maybe making a wiki with explanations for all the values? like, 3b-3f looks like cell voltages,  1c is the serial number, etc... i was going to read some pdfs again these days that start with sluu, but wouldn't it be nice to have this information in a more human readable format, without all the scrolling, like a simple table maybe? I was able to capture the communication with my dead but still talking ideapad battery, so i know what it will be asking for. 
The ability to simply make up certain values seems very practical for testing! 
I was wondering, how do you tell the system that a battery is present, when it's not? resistor between + and - should do i guess?
And how about adding a temperature sensor instead of hardcoding 22,1 ºC, i mean, it's an arduino? And maybe add some logic that doesn't charge when the temperature is too high?

iam4722202468, 2021-01-18:
Hey!
Yeah, this is basically an implementation of the SMBus for an attiny controller. Thanks for letting me know about the crashing, about a week ago I was thinking about this code and realized that crashing might have been a problem when it got requests that weren't defined, but wasn't sure.

> I was wondering, how do you tell the system that a battery is present, when it's not? resistor between + and - should do i guess (edit: seems like a 10kohm resistor between the thermistor and ground does the trick.)

Sounds right. Interestingly, I noticed that the cheap third party battery I took apart uses a 10k resistor on the temperature sensor pin too.

> And how about adding a temperature sensor instead of hardcoding 22,1 ºC, i mean, it's an arduino? And maybe add some logic that doesn't charge when the temperature is too high?

Another good idea, it would definitely be possible, would be easy to add, and there's free pins that aren't used for anything on the attiny right now.

The long term plan is to make this an open platform for all old batteries, where there's one pcb that can be programmed to work with any laptop. Right now I'm taking a bit of a break from this project while I wait on a bunch of attiny pcb's to be manufactured, once those come in I'll continue doing more testing. Have learned a lot about this since I started the project, hopefully I'll be able to expand on what I've already posted soon.

5ch4um1, 2021-01-19:
Well, also not limited to attiny, i tried with an arduino nano clone and it worked, and i could imagine that this will work on a lot of other platforms too ;)
And for the temperature sensor, "A thermistor is a temperature sensitive resistor..." so 10kohm means something around 21º C iirc. Would probably be possible to recycle one from an old battery pack and use it with the microcontroller. 
I wonder if the temperature is measured by both the battery and the laptop? I mean, the battery for sure, but the laptop has that pin, so i guess it somehow uses it?
Anyway, i'd definitely recommend to keep a simple version around as the project grows, for educational and testing purposes it seems very practical as it is at the moment. 
speaking about old batteries, do newer devices have some kind of authentication mechanism for normal operation mode?
Newer devices seem to have secure auth to get them into unsealed or full access mode, but i haven't heard of anything like this for normal operation mode?
Anyway, there is a huge potential in this, this is literally power to the people! 
thank you very much.

iam4722202468, 2021-01-20:
> do newer devices have some kind of authentication mechanism for normal operation mode
Yeah, this is a bit of a problem, quite a few devices require the correct reply from the auth command. The T430 actually has an auth command, where an incorrect reply will stop the battery from charging.

One of the benefits on using an attiny instead of a conventional SMbus IC is that they're easy to program, and in the future if exploits or workarounds are found for the auth sequences it's super easy to update and test the code. Some manufactures also implement custom commands, and again for the same reasons using an attiny here can simplify things.

TheWorldIsBetterWithElectronics, 2021-07-23:
Hello,
This is a very interesting project. You did a hell of a job. Thanks you. But could you reediter the code with a little more comments so that it is more understandable because I am not a code genius. Because I would like to modify the project so that it adapts to my computer but I quickly get lost in all these lines. Thank you if you have some time to help me to understand the code.

iam4722202468, 2021-07-23:
@TheWorldIsBetterWithElectronics 
> Hello,
> This is a very interesting project. You did a hell of a job. Thanks you. But could you reediter the code with a little more comments so that it is more understandable because I am not a code [...]

Thanks for the feedback! I added some comments to explain what's going on in the code, hopefully that helps you.
For future reference, it's generally better to create a separate issue for new requests as it makes it easier for the developer to find the message and helps you more easily track work done.

TheWorldIsBetterWithElectronics, 2021-07-23:
Thanks for the improved code feedback, it's still easier to understand the big parts of the code now. I will continue to experiment with the battery of my computer and I will tell you if it works or not. I will probably come back to ask questions but now, I will not forget to create a new  issue . Thank you very much.