iam4722202468/ThinkpadBattery #5: gap between request and answer?

5ch4um1, 2020-12-26:
Not a real issue, seems to work anyway, i just noticed that the arduino takes a bit longer to answer (roughly 420 microseconds) than the original battery, here the arduino (nano clone):
[image]

and here the original: 

[image]
is that maybe the time it takes to calculate the pec, or why is this happening?

digmorepaka, 2020-12-27:
Not sure if it's related, but apparently if you have the Arduino bootloader flashed the atmels tend to be slower in certain operations, does this occur when flashing the compiled binary over icsp while ommiting Arduino bootloader etc

5ch4um1, 2020-12-27:
you mean cross compiling it with avr-gcc and flashing it with avrdude? no, haven't tried this yet. and actually this looks a bit complicated for the newbie that i am. and again, it works anyway, the delay seems to be within what's acceptable for the laptop. 
I think i have like a million questions, is there maybe a thread in a forum, subreddit, discord, or anything like that? 
Anyay, i tested today with my Ideapad G550 and so far it seems to work pretty much "out of the box", though it won't accept any other vendor name than the original, defaults to COMPAL otherwise: 
[image]

iam4722202468, 2021-01-18:
Haven't done any testing for this yet, but if the delays are consistently the same length for all the packets I suspect the attiny is just taking a while to process the input and create the output.

iam4722202468, 2021-01-19:
@5ch4um1 if you want to ask questions, I'm on the thinkpad discord with a bunch of other people who would be happy to answer them! https://discord.com/invite/Ybdz7AS

5ch4um1, 2021-01-19:
Thanks again, very much appreciated.
And i don't know, i tested with an arduino nano clone, could test now again with a wemos d1 mini, don't have a second logic analyzer to see if maybe something was wrong with that part of my setup, but as mentioned, it works anyway. I may or may not have posted this issue just to point out that the gap was 420 microseconds... ;)
Closing for now, will reopen if it turns out to be a real functional issue.

closed 2021-01-19 by 5ch4um1