iam4722202468/ThinkpadBattery #6: Help me please. The voltage of 12367mV is displayed all the time, it always shows 100% of the charge

Naughtydog2004, 2021-02-08:
Hello. Can you please tell me which command controls the battery state of charge at 100% and charge / discharge mode? I can manage charge percentages with the byte reply0x0D command. But my laptop still shows 100% and won't charge the battery. By the way, the voltage from the resistor divider also does not change in any way and is transmitted to the SMBus as 12367mV (although, separately from this program, the code for measuring the voltage functions properly through the analog pin A3 on the arduino nano). Thanks in advance.
https://drive.google.com/file/d/1x5DdUYXAxN2JYD1_1s9keKdGmnjLt5Kj/view?usp=sharing

iam4722202468, 2021-02-10:
Looking at my code, it looks like reply0x0F is what handles the % charged, try manually setting the charge value to something and checking if that changes the charge of your device. If it doesn't, then your device might need other packets than just the 0x0D one. Not all implementations of the smbus are identical for all devices

iam4722202468, 2021-02-10:
As for the voltage divider, there's a chance I screwed something up and it doesn't actually work, really hoping that isn't the case though... either way, i'll probably redo a few parts of this in around 4 months from now. The idea is to make a small pcb with a 3s bms and attiny combined, that can be programmed for any device.

Naughtydog2004, 2021-02-10:
Thank you! If I can fix this, then I will inform you

luketimothyjones, 2022-05-13:
I think `0x0f()` is incorrect, unless the Thinkpad battery departs from the SMBus spec (which is entirely possible). Command `0x0f` is `RemainingCapacity()`, which is expressed as mAh or mWh depending on the value of `0x03` bit 15 (aka `CAPACITY_MODE`): `0` for mAh and `1` for capacity/10 mW/mWh. The calculation in `0x0f` is identical to `0x09` (`Voltage()`)above it, perhaps it was a copying mishap?

iam4722202468, 2022-05-14:
@luketimothyjones yes this does look like an issue. When making this I didn't really know what I was doing, and it looks like I did just copy it. Feel free to pr a fix