iam4722202468/ThinkpadBattery #11: SMBUS Communication Stops Wehn Grabing Battery Vender on Emulated Battery

Sam64-png, 2021-11-22:
Hi, I am trying to power a A485 think pad from my lab bench power supply and uploaded this code onto an Arduino Nano. I have it all wired correctly and it successfully communicates with the Arduino. I changed the battery reference voltage to 5 volts, set the analog pin to A3 and A3 is reading around 4.4 volts. When powering the laptop it never turns on but some of the i2c commands and found that it never gets to the point were it reads the voltage but stops once it reads the vender. Here is a log of all the commands sent.
`0x03 0x15 0x18 0x19 0x1A 0x1B 0x1C 0x20`. I kept the vender in the code to the default value "HACKED". I do not have a good way to test the laptop on a working battery. What happened is that its charging circuitry got destroyed to I had to desolder some parts for it to come on. Now when i plug in a almost dead battery it turns on but dies a few seconds later.

I have to thank you a lot seance this code is a life savor seance i don't want to spend a lot of money to get a new one and I don't like throwing out a whole computer when only one part fails. I really hope you can help. Once i get this done i plan to make a custom pc with it and a type c connector and battery to get a lot of the old functionality back.

iam4722202468, 2021-11-24:
Please keep in mind that batteries are dangerous, and this project is meant to be used with a bms, it is not a replacement for a bms. If you're directly connecting the cells to the circuit that could be the problem

Sam64-png, 2021-12-01:
> 

Sorry I probably did not explain it too well. I was using the project without any battery so I could power the laptop from a bench power supply threw the battery connector. I tied the battery voltage detection analog pin to a adjustable voltage reference for testing.  Also in the mean time I am assuming the laptop stops communicating with the battery because it may be checking the vender id to stop agent cheaper off brand batteries. I been trying to get the vender id from a preexisting think-pad battery using the same commands from the code. It responds to that IC2 address but always replys with 0xFF. I don't have a logic analyzer directly intercept its communications so i may have to wire something up there. If you know what is wrong in the arduino code causing this setup or model of laptop to mess up any help is greatly appreciated.

iam4722202468, 2022-01-04:
@Sam64-png Sorry for taking so long to reply!

You are correct in saying that some batteries expect verification codes, it looks like the A485 model came out recently and most laptops newer than the T420 series do have verification. The verification will not be possible to get around without some exploit or a modified EC firmware, and I don't think either of those exist for that laptop (yet).

The reason you can't simply send the same responses is because the motherboard most likely sends the battery a challenge-response test, which involves a message being sent from the motherboard to the battery, the battery signing the message with a private key, and the motherboard verifying the signature with a public key. Without the private key there's no way to get the battery to charge

I'm sorry to hear that your charging circuitry got destroyed, that's unfortunate.

There's a chance that the program is crashing, this can happen with the motherboard sends a packet type that isn't recognized. You could check this by using an arduino with serial output like an arduino uno, modifying the code to log the packets that it receives, running the code and checking if there are packets sent that don't appear in the code. That being said, even if you get a little bit more communication working between the battery and the motherboard you will never be able to get past the auth sequence.

Good luck with your future projects!

Sam64-png, 2022-01-04:
> @Sam64-png Sorry for taking so long to reply!
>
> You are correct in saying that some batteries expect verification codes, it looks like the A485 model came out recently and most laptops newer than the [...]

Alright sounds good. I expected something in that case seance I felt they would of changed that later on so people buy the more expensive batteries. I went ahead and got a new laptop. Idk if I trust the chargers or at least the old one I had because i am wondering if that is what caused the issue with the charging in the first place. I had one break a while back and never investigated why it was broken seance their assembled into a solid brick, but I wonder if it was caused by the capacitors going bad and it is generating an unfiltered output with voltage spikes. Thank you for the help.

closed 2022-01-04 by Sam64-png

luketimothyjones, 2022-05-02:
> There's a chance that the program is crashing, this can happen with the motherboard sends a packet type that isn't recognized.

For historical reference, this was resolved by #12