iam4722202468/ThinkpadBattery #14: PLEASE MAKE FOR LENOVO X220

jakevin99, 2022-09-24:
I am planning to recell my 9 cell x220 battery but problem is the chip inside.

iam4722202468, 2022-10-07:
This will probably work on the x220 without any changes

iam4722202468, 2023-11-27:
> But will it work on **30 models?

Nah, unmodified the **30 models use a signature packet between the battery and laptop to ensure the laptop is authentic. No way around this, as message replies cannot be validated without having the key. There is [this](https://github.com/hamishcoleman/thinkpad-ec) project which you can use to remove the signing and make the battery work.

Michal-Szczepaniak, 2023-11-27:
What if i have corebooted **30 model? Then it will work?

iam4722202468, 2023-11-27:
Coreboot itself is only a modification for the bios. The EC (Embedded Controller) is what communicates with the battery, and is what needs to be modified.

Michal-Szczepaniak, 2023-11-27:
gotcha

Bromiun, 2023-11-28:
Isn't there a practical way to get around this problem?  A pack is usually 3S2P.  And let's say you have 6x 18650s to replace.  

Charge the pack and 6 good cells to full.  Solder (with wires) 3S of the good cells to the existing two power strips and two balance strips in the pack.  This will keep the handshake trap happy.

Remove 3S of the bad cells.  Replace with good cells.  Either soldered or spot welded.

Remove the 2nd set of 3S bad cells and replace with the first 3S of good cells.

I believe that it is sufficient for one cell to go bad, to make the battery pack useless.  So test the 6 bad cells and you may find a few still operational.

iam4722202468, 2023-11-28:
That is possible, but it requires having a working battery. The goal of this project is to create a working battery from scratch.

Bromiun, 2023-11-29:
Yes, a valid point.  During my 15 years having multiple ThinkPads, I have accumulated a lot of packs with bad cells.  Buying packs off eBay is a possibility and my last purchase I believe about $50 is fantastic.  I used to open the packs up and replace the cells straight.  I soon discovered that is a good way to destroy a pack.

At one time, the laptop male connector got misaligned, and due to a defect in the design, when you push in the battery, it even clicks giving you a false sense that connection has been made. I could not get any pack to work and there was no mention of this issue in any forum.  I went through 3 or 4 packs the discover the problem and simple aligned the connector and it has been working to this day.. So I have a few working packs.  The design problem is that the female connector on the pack is spring loaded.  So it does not have to mate with the male terminal to insert the pack, and you can still push it in and force a false positive click.

I am sure you can find cheap packs that have bad cells but are still alive.  Have you tried this?