iam4722202468/ThinkpadBattery #15: Arduino Uno pack

Bromiun, 2022-09-24:
Hi Iam, I am new to Github, so am not sure how to connect with you.  This may not be the correct manner.  Please let me know.

I want to connect my 3s5p 2170 pack to my T540.  This will be in a box along with an Uno, separate from the laptop.  There will be no PCB.  There may be a 2x40 LCD to report on usage, V, Wh, etc.  No need to inform the T540 on the status.  BMS and protection and charging all in the box.  This will be the simplest method to defeat Lenovo's anti-consumer lock.

I need to figure out how to connect the Uno to the Thinkpad.  I am assuming that after the handshake, the Uno can can go to sleep?

If I connect the I2C bus, and the Gnd and Vcc to B+ and B-, download your software, will this do the job?  In which case I only need a female connector, or even could hardwire that.

Before I start this project, I need your blessing and help.  Thank you so much.

Brom
Brom dot Nader at gmail dot com

iam4722202468, 2022-10-07:
Hey,

Yes, github is a great way to connect with project owners (: I only check this project occasionally, but I will eventually see issues posted here.

The T540 batteries use a key signing process that can't be defeated without the private keys on the battery, which as far as I know nobody has access to. This means that unfortunately, this project won't work with it. This is the case for anything newer than the **20 series, with the **30 series having a project that lets you rewrite the embedded controller found [here](https://github.com/hamishcoleman/thinkpad-ec)

With an older, supported laptop (anything before and including the x220, T420, etc.), the project would be connected to the battery connector. The battery itself uses I2C over the connector pins, like this
[image]

When the laptop is on, it's constantly communicating with the battery for information like remaining capacity, voltage level, temperature, etc. and from what I've seen, if these packets aren't received or the checksum is wrong, the laptop will go through the handshake process again.

Bromiun, 2022-10-25:
> 
Thank you for your reply, which I just have seen. As you say, the private key (of laptop or pack?) is in the pack.  So how do the Chinese clone makers manage to make clones?

Also, my female connector leads look different than the image you have.

BTW, a couple years back, for a long time, the W540 refused to recognize legitimate packs.  I kept on ordering new clones, and still they were not recognized.  After spending a lot of time on this, I realized the problem was that the male and female connectors did not line up and mate exactly, as the male leads were bent, so there was no proper contact.  You would think that if this happens, the pack should not click in place and laptop would resist insertion of the pack.  Turns out the female connector on the PCB in the pack has no backing.  You can easily push the female connector in, so no mating was necessary for the pack to click in!  Imagine if I had ordered expensive Lenovo packs, and they would still have this problem.  Very bad design.

 I had a 3S3P clone battery, with only 10 minutes of juice remaining for the Lenovo W540.  So I opened it up.  Soldiered 3S to the 4 leads to keep the power on, and then carefully removed the 9 cells.  The cover said 7.8 Ah.  But each cell says 2.2 Ah, so a total of 6.6 Ah.  Quite a scam.  Moreover,the cells were Chinese made, even though the model number was like a Korean cell.  Most of the cells tested at 0.7 Ah, and some as low as 0.12 Ah, even though they could hold a 4V charge.

After building a 3S3P with Panasonic NCR18650B (3.4 Ah), I carefully inserted cells and soldiered the leads.  In this process one of the two balance leads got detached and lost its power for a short while.

Now the W540 recognizes that there is a pack there, but refuses to charge it and Windows shows a black cell icon.  I rebalanced the cells but that made no difference.  When I disconnect power, the laptop remains on for 20 seconds, and then it looks like it is going to sleep with gradual dimming.  But does not awake.  When power is connected, it reboots from scratch.

iam4722202468, 2022-11-06:
I'm not sure how the Chinese clones work, but for newer models they must have somehow gotten the key from an authentic battery pack, which isn't possible for us to do.

Recelling packs is definitely a good choice, although I'm not sure what exactly happened to your pack. You could check the laptop on Linux and see if has the same behavior, but my guess is there's an issue with the pack's bms which probably isn't solvable.

auroraanna, 2025-01-08:
if working battery packs have the private key on a flash chip separate from their microcontroller, you can probably sniff on the pins between the microcontroller and the flash chip for the private key